WLAN-Pi/wlanpi-core #9: API For Wi-Fi Channel Scanning

JudahZF, 2024-09-02:
HI There,

I'm building some software for Wi-Fi monitoring at events, and the wlan pi looks like a great tool.
Will there ever be an API to get the results of a Wi-Fi scan. I understand the data goes into Grafana, but I can't find any way of getting it out and into another system (although I may have missed it). I am happy to work on a PR to add this if it seems like a good idea? Would it be best placed here with the API or in the Grafana repo with the nlscan app?

Thanks,
Judah

joshschmelzle, 2024-09-10:
Hey Judah, today, the wlanpi-core API services are not exposed and are currently only being used for RPC between the frontend and backend. I think the right approach would be some (new) API service similar to wlanpi-core, but on an exposed port with some sort of self service token management. I don't think that the grafana repo is the right place for this.

joshschmelzle, 2024-09-11:
We are also already working on changes to allow the WebUI to get and display scan results.

JudahZF, 2024-09-16:
Hi Josh,
Thanks for the reply. I think JSON data would be the ideal, but I would probably be able to work with most types of data transfer, as I am building individual services to handle the data from each system I plan on working with.
Would the best approach be to start working on something like this myself that can then be an add on or moved into wlan pi, or wait for the project to have a repo / idea of how this would work and then contribute?

joshschmelzle, 2024-09-16:
I would wait so that we don't duplicate efforts with different implementations.

joshschmelzle, 2024-09-29:
@JudahZF the latest version of core has scanning capabilities. 

Usage examples:

```
curl -X 'GET' \
  'http://<IP>:8000/api/v1/network/wlan/scan?type=active&interface=<INTERFACE>&timeout=20' \
  -H 'accept: application/json'
```

```
http://<IP>:<PORT>/api/v1/network/wlan/scan?type=active&interface=<INTERFACE>&timeout=20
```

Output response example:

```
{
    "nets": [
        {
            "ssid": "SSID_1",
            "bssid": "02:d1:37:56:78:9a",
            "key_mgmt": "wpa-eap/wpa-eap-sha256",
            "signal": -44,
            "freq": 5580,
            "minrate": 6000000
        },
        {
            "ssid": "SSID_2",
            "bssid": "02:4d:7b:9c:3d:11",
            "key_mgmt": "wpa-psk/sae",
            "signal": -45,
            "freq": 5180,
            "minrate": 6000000
        },
        {
            "ssid": "SSID_3",
            "bssid": "02:3e:8c:25:bd:2f",
            "key_mgmt": "wpa-psk/sae",
            "signal": -39,
            "freq": 5180,
            "minrate": 24000000
        },
        {
            "ssid": "SSID_4",
            "bssid": "02:17:fc:91:2a:7e",
            "key_mgmt": "wpa-psk/sae",
            "signal": -45,
            "freq": 5180,
            "minrate": 6000000
        },
        {
            "ssid": "SSID_5",
            "bssid": "02:5b:ac:16:34:9f",
            "key_mgmt": "sae",
            "signal": -47,
            "freq": 5620,
            "minrate": 24000000
        },
        {
            "ssid": "SSID_6",
            "bssid": "02:e6:39:7c:12:8a",
            "key_mgmt": "wpa-eap-suite-b-192",
            "signal": -47,
            "freq": 5620,
            "minrate": 24000000
        },
        {
            "ssid": "SSID_7",
            "bssid": "02:bc:df:3e:5f:73",
[... 257 lines cut ...]
            "signal": -77,
            "freq": 2437,
            "minrate": 11000000
        },
        {
            "ssid": "SSID_40",
            "bssid": "02:62:5f:7e:45:2c",
            "key_mgmt": "wpa-psk",
            "signal": -88,
            "freq": 2437,
            "minrate": 1000000
        },
        {
            "ssid": "SSID_41",
            "bssid": "02:9e:3f:6e:4c:8b",
            "key_mgmt": "wpa-psk",
            "signal": -90,
            "freq": 2437,
            "minrate": 1000000
        }
    ]
}
```

What changes might you want to see in the output?

JudahZF, 2024-09-29:
Looks good. It would be useful to be able to see the Channel width as well if possible, as this would give a good environment overview.

joshschmelzle, 2024-10-04:
@bentumbler @Orbitix can either of you handle tackling adding channel width to the v1/network/wlan/scan endpoint response?

bentumbler, 2024-10-07:
yes.  I'll schedule this in during this week

joshschmelzle, 2026-09-18:
Done: canonical GET /api/v1/utils/wlan/scan (utils_api.py) backed by wlanpi_core/wlan/scan.py and schemas/utils/wlan_scan.py, covered by tests/test_wlan_scan.py and test_wlan_scan_api.py.

closed 2026-09-18 by joshschmelzle

JudahZF, 2026-09-22:
Thanks guys!