WLAN-Pi/wlanpi-core #12: Fix code scanning alert - Uncontrolled command line

joshschmelzle, 2024-09-29:
Tracking issue for:
- [ ] https://github.com/WLAN-Pi/wlanpi-core/security/code-scanning/21

joshschmelzle, 2024-09-29:
Affected code was added in #11. We may need to consider how to prevent abuse of this code. @MichaelKetchel

joshschmelzle, 2024-10-07:
We've all contributed code that uses subprocess. We need to have a hardened approach that works for all of us.

MichaelKetchel, 2024-10-11:
Looking at this--you mentioned an allowlist, but I'm not sure what allowlist we're talking about. Maybe there's something in https://github.com/WLAN-Pi/wlanpi-core/security/code-scanning/21 that i can't see--when i click it I get a not found.

In general, I'd like to see a standardized, hardened approach for subprocess usage for sure. My function in [wlanpi-core/wlanpi_core/utils/general.py](https://github.com/WLAN-Pi/wlanpi-core/blob/f2183cd4c1faaf5a62833deed6e9a7aba7221cb1/wlanpi_core/utils/general.py#L7-L19) mentioned above was something of a proposal for a more flexible way to handle subprocess calls (being able to always get both stderr and stdresult, a nice json formatter, and raising on command fail by default to avoid silent failures. I need to make a version that handles async.

I think there's a few cases where needing shell capabilities exist (currently), notably piping things between processes, but in reality these can probably be sidestepped by creating "wrappers" for commands like `jc` and `grep` that are used super frequently in the piping and a pattern to ease piping output of a regular run_command or run_cli_async to them using subprocess pipes and enforce always using the safe array usage.

Do you have any strong opinions on where these things should live? For whatever reason, our utilities/helpers living under "services" feels misleading, as they're used in some other places--that's why I created a utils folder, but maybe that's not ideal either.

When we figure this out, I can do a refactoring pass and make everything unified.

MichaelKetchel, 2024-10-25:
My proposal for a unified synchronous and async interface for running shell commands

```python

def run_command(cmd: Union[list, str], input:Optional[str]=None, stdin:Optional[TextIO]=None, shell=False, raise_on_fail=True) -> CommandResult:
    """Run a single CLI command with subprocess and returns the output"""
    
    # cannot have both input and STDIN, unless stdin is the constant for PIPE or /dev/null
    if input and stdin and not isinstance(stdin, int):
        raise RunCommandError(error_msg="You cannot use both 'input' and 'stdin' on the same call.", status_code=-1)
    
    if shell:
        cmd: str
        logging.getLogger().warning(f"Command {cmd} being run as a shell script. This could present"
                                    f"an injection vulnerability. Consider whether you really need to do this.")
    else:
        cmd: list[str]
    cp = subprocess.run(
        cmd,
        input=input,
        stdin=stdin,
        encoding="utf-8",
        shell=shell,
        check=False,
        capture_output=True,
    )

    if raise_on_fail and cp.returncode != 0:
        raise RunCommandError(cp.stderr, cp.returncode)
    return CommandResult(cp.stdout, cp.stderr, cp.returncode)

async def run_command_async(cmd: Union[list, str], input:Optional[str]=None, stdin:Optional[TextIO]=None, shell=False, raise_on_fail=True) -> CommandResult:
    """Run a single CLI command with asyncio.subprocess and returns the output"""

    # cannot have both input and STDIN, unless stdin is the constant for PIPE or /dev/null
    if input and stdin and not isinstance(stdin, int):
        raise RunCommandError(error_msg="You cannot use both 'input' and 'stdin' on the same call.", status_code=-1)

    # asyncio.subprocess has different commands for shell and no shell.
    # Switch between them to keep a standard interface.
    if shell:
        cmd: str
        logging.getLogger().warning(f"Command {cmd} being run as a shell script. This could present"
                                    f"an injection vulnerability. Consider whether you really need to do this.")

        with asyncio.subprocess.create_subprocess_shell(
                cmd,
                stdin=asyncio.subprocess.PIPE if input else stdin,
                stdout=asyncio.subprocess.PIPE,
                stderr=asyncio.subprocess.PIPE
        ) as proc:
            proc: Process
            stdout, stderr = await proc.communicate(input=input.encode() if input else None)
    else:
        cmd: list[str]
        with asyncio.subprocess.create_subprocess_exec(
                cmd[0],
                *cmd[1:],
                stdin=asyncio.subprocess.PIPE if input else stdin,
                stdout=asyncio.subprocess.PIPE,
                stderr=asyncio.subprocess.PIPE
        ) as proc:
            proc: Process
            stdout, stderr = await proc.communicate(input=input.encode() if input else None)

    if raise_on_fail and proc.returncode != 0:
        raise RunCommandError(error_msg=stderr.decode(), status_code=proc.returncode)
    return CommandResult(stdout.decode(), stderr.decode(), proc.returncode)
    ```

joshschmelzle, 2024-10-25:
This looks good to me. I think it makes sense to put these at /utils/general.py 

The /services/helpers.py should be refactored to remove run_cli_async() and run_command(), and possibly named more appropriately since it's just holding constants for now.

MichaelKetchel, 2024-10-25:
That was my plan exactly! Will carry on!