WLAN-Pi/wlanpi-core #45: [Bug]: Missing security on API

joshschmelzle, 2024-10-24:
wlanpi-core is missing security.

also consinder adding a rate limiter (such as `fastapi-limiter`).

joshschmelzle, 2024-10-27:
Related PRs: #51 #52

Orbitix, 2024-10-28:
I have created pull request #56, which addresses the authorization aspect of the api, adding a JWT token requirement to the endpoints, which can be generated only locally on the pi via an endpoint, and not externally.

This is a proposal for a system, and it is open to discussion on how to handle the system and also how to address the breaking of the webui, as it currently does not support the token system.

There also needs to be a script on the pi that allows for easy token generation by the logged in user, and also a discussion around how to handle secret keys for the token generation.

joshschmelzle, 2025-01-13:
#56 is merged. Usage examples to follow.

joshschmelzle, 2026-09-18:
Done: API auth is enforced via verify_auth_wrapper in core/auth.py (JWT bearer + localhost HMAC) applied across all routers with Depends(), covered by tests/test_auth.py and tests/test_auth_token_api.py.

closed 2026-09-18 by joshschmelzle