WLAN-Pi/wlanpi-core #62: [FEAT] Logging

joshschmelzle, 2024-11-04:
Issue to improve logging throughout core. Related PR is #26.

Proposing to consider Pr #26 switching to structlog. 

- https://www.structlog.org/en/stable/
- https://gist.github.com/nymous/f138c7f06062b7c43c060bf03759c29e

Consider 3 levels per Ben's comment here https://github.com/WLAN-Pi/wlanpi-core/pull/26#issuecomment-2386059907.

A nice to have would be a way to bring logs into grafana.

joshschmelzle, 2025-01-14:
#26 has been merged which has quite a few changes for logging.

bentumbler, 2025-01-15:
A good idea for the backlog but as we are unlikely to implement anything to consume the logs anytime soon shall we just freeze where we are now for continued testing and pick this back up after WLPC. 
Significant testing of the api starts on Saturday so if you do want to do something sooner then maybe target cop Friday for feature freeze

joshschmelzle, 2026-11-05:
The structured logging from #26 is merged and `app.log` uses the JSON formatter. The remaining concrete work is tracked in #379 (log rotation) and #380 (debug.log record size).

closed 2026-11-05 by joshschmelzle