WLAN-Pi/wlanpi-core #70: [BUG]: rate limiting 504s

joshschmelzle, 2025-01-14:
### What happened?

Need to investigate scenarios that cause 504 timeouts.

### Relevant log output

```shell
N/A
```

### How often does this bug happen?

Sometimes

### What version are you using?

2.0.0

### Self Service

- [X] I would be willing to fix this bug myself.

### Guidelines and Policies

- [X] I have read the contributing guidelines and agree to follow the code of conduct and contribution policies.

joshschmelzle, 2025-01-16:
Title change because rate limiting with slowapi is not the issue after some testing.

Orbitix, 2025-01-19:
I am having this issue whilst testing the app, and getting 504 timeouts.

Here are logs i found in the journal:

```
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]: 2025-01-19 23:19:20,219 - wlanpi_core.core.database - ERROR - Database connection error
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]: Traceback (most recent call last):
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:   File "/opt/wlanpi-core/lib/python3.9/site-packages/wlanpi_core/wlanpi_core/core/database.py", line 303, in get_connection
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:     yield connection
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:   File "/opt/wlanpi-core/lib/python3.9/site-packages/wlanpi_core/wlanpi_core/core/auth.py", line 1299, in purge_expired_tokens
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:     cursor.execute(
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]: sqlite3.IntegrityError: FOREIGN KEY constraint failed
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]: 2025-01-19 23:19:20,239 - wlanpi_core.core.auth - ERROR - Failed to purge tokens: Could not connect to database: FOREIGN KEY constraint failed
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]: Traceback (most recent call last):
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:   File "/opt/wlanpi-core/lib/python3.9/site-packages/wlanpi_core/wlanpi_core/core/database.py", line 303, in get_connection
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:     yield connection
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:   File "/opt/wlanpi-core/lib/python3.9/site-packages/wlanpi_core/wlanpi_core/core/auth.py", line 1299, in purge_expired_tokens
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:     cursor.execute(
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]: sqlite3.IntegrityError: FOREIGN KEY constraint failed
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]: During handling of the above exception, another exception occurred:
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]: Traceback (most recent call last):
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:   File "/opt/wlanpi-core/lib/python3.9/site-packages/wlanpi_core/wlanpi_core/core/auth.py", line 1310, in purge_expired_tokens
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:     log.debug(
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:   File "/usr/lib/python3.9/contextlib.py", line 193, in __aexit__
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:     await self.gen.athrow(typ, value, traceback)
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:   File "/opt/wlanpi-core/lib/python3.9/site-packages/wlanpi_core/wlanpi_core/core/database.py", line 342, in get_connection
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]:     raise DatabaseError(f"Could not connect to database: {e}")
Jan 19 23:19:20 wlanpi-bc2 gunicorn[4805]: wlanpi_core.core.database.DatabaseError: Could not connect to database: FOREIGN KEY constraint failed
```