WLAN-Pi/wlanpi-core #73: [FEATURE]: scandump/tshark wrapper

joshschmelzle, 2025-01-17:
### What is the feature?

`sudo scandump wlan0 -c 1 - | tshark -r - -T json` for everything. 

`sudo scandump wlan0 -c 1 - | tshark -r - -T json -e wlan.bssid` for just BSSIDs.

### Motivation

WLAN Pi Go

### Self Service

- [x] I would be willing to fix this feature myself.

### Guidelines and Policies

- [x] I have read the contributing guidelines and agree to follow the code of conduct and contribution policies.

joshschmelzle, 2025-01-17:
Unless easily getting advanced details from existing work.

adriangranados, 2025-02-28:
For this endpoint to be used with an app like WFE, it would need to provide each frame as raw data. A base64 encoded string would work.

joshschmelzle, 2025-03-21:
@adriangranados @Orbitix thoughts on how you want to receive this data? SSE? Websockets? gRPC? SSE might make sense because comms for this really only need to be from server to client.

adriangranados, 2025-03-21:
I think SSE should be okay. Would it help if scandump could output in JSON format (base64 encoded), for example?

joshschmelzle, 2025-04-11:
Note to provide a way to scan by band or group of bands or all bands.

joshschmelzle, 2026-10-20:
The Grafana Scanner dashboard is another consumer. WLAN-Pi/wlanpi-grafana#44 moves it from the bundled `nlscan-lp` binary to the one-shot scan API, called every 15 s per interface. It would switch to a continuous stream from here once one exists. What it needs:

- Per-interface scanning, with the interface found in any namespace (as the one-shot API already does), so several adapters can stream at once.
- Per BSS: BSSID, signal, frequency, primary channel, and the raw SSID bytes (or decoded UTF-8; see #400).
- SSE delivery. Parsed fields per BSS would do for Grafana; WFE's base64 frames can sit alongside them.
- Coordination with the one-shot scan lock. Today a WebUI scan and the Grafana stream get 409 `SCAN_IN_PROGRESS` from each other; with scandump as the engine, one-shot scans could be served from the running stream instead of racing it.