WLAN-Pi/wlanpi-core #85: [BUG]: /api/v1/network/wlan/scan always return {type} is not a valid scan type

zaffo80, 2025-03-31:
### What happened?

I tried to run /api/v1/network/wlan/scan from the API documentation or Postman, but it always return "active is not a valid scan type" or "passive is not a valid scan type". I tried to use type active or  passive + interface wlan0 + timeout 20.

I tried to add a row with "else:" before the last row of the procedure "async def get_async_systemd_network_scan" in the network_service.py file, but I receive an Internal Error message.

### Relevant log output

```shell
Internal Server Error
```

### How often does this bug happen?

Every time

### What version are you using?

3.3.0

### Where is the bug located?

/opt/wlanpi-core/workingdirectory/services/network_service.py

### Self Service

- [ ] I would be willing to fix this bug myself.

### Guidelines and Policies

- [x] I have read the contributing guidelines and agree to follow the code of conduct and contribution policies.

zaffo80, 2025-03-31:
I tried to look for any logs on /var/logs/, but I cannot find anything useful to solve this issue.

joshschmelzle, 2025-04-01:
This is in fact a bug. We are not happy with the current implementation to get scan results and are working on a way to improve the service that feeds this API.

joshschmelzle, 2026-09-18:
Done: the canonical wlan/scan endpoint validates the scan detail via wpa_scan.normalize_scan_detail (returns 400 on invalid input, asserted in tests/test_wlan_scan_api.py), and the deprecated /network/wlan/scan route neutralizes the unused type param.

closed 2026-09-18 by joshschmelzle