WLAN-Pi/wlanpi-core #89: [FEATURE]: Profiler API support

Orbitix, 2025-05-20:
### What is the feature?

Allows controlling Profiler through the api, with support for start arguments and in the future getting results.

### Motivation

This is needed so that the App can control Profiler as part of expanding the capability of the App

### Self Service

- [x] I would be willing to fix this feature myself.

### Guidelines and Policies

- [x] I have read the contributing guidelines and agree to follow the code of conduct and contribution policies.

joshschmelzle, 2025-06-09:
#90 merged.

joshschmelzle, 2026-09-18:
Done: profiler API shipped in wlanpi_core/api/api_v1/endpoints/profiler_api.py (GET /status, POST /start, /stop, auth-protected) with schemas in profiler/schemas.py and tests/test_profiler_process.py.

closed 2026-09-18 by joshschmelzle