WLAN-Pi/wlanpi-core #91: [FEATURE]: WebSocket integration for data streaming

Orbitix, 2025-05-20:
### What is the feature?

WebSockets will be used for features such as AP finder, spectrum analyser,  scans and more. it will allow for the user to change parameters or settings in real time and receive the data on the same channel.

### Motivation

Required for certain features mentioned above, as they need continuous data streaming and command+control

### Self Service

- [ ] I would be willing to fix this feature myself.

### Guidelines and Policies

- [x] I have read the contributing guidelines and agree to follow the code of conduct and contribution policies.

joshschmelzle, 2025-06-09:
#90 merged.

joshschmelzle, 2026-09-18:
Done: WebSocket capture streaming in wlanpi_core/api/api_v1/endpoints/streaming_api.py (WS /api/v1/streaming/capture with first-message JWT auth, sessions/subscribers), wlanpi_core/streaming/ module, covered by test_streaming_api.py, test_streaming_processes.py, test_capture_ws_auth.py.

closed 2026-09-18 by joshschmelzle