WLAN-Pi/wlanpi-core #98: [FEATURE]: Improved client connection management with interface isolation

bentumbler, 2025-07-16:
### What is the feature?

Connect to Wi-Fi SSIDs from one or more adapters from a centralised API, which allows all interfaces to be used simultaneously for various purposes such as testing.

The feature should specifically allow a wlan interface to be connected as a client to perform tests while the ethernet interface or another wlan interface interface remains active for telemetry

### Motivation

### Why do anything?
An initial implementation of Wi-Fi client connection control via DBUS has proven to have some limitations such as a need to re-map the systemd service to allow DBUS control and some limitation in accessing all the features of the wpasupplicant. It allows for the WLANPi to be left in a poor state which is confiusing. Currently persistence of wlan configurations is not implemented.

### An updated approach is proposed
Work done by others identified that through the use of namespaces, each interface can be moved in and out of isolated scopes or 'namespaces'. This allows a wlan adapter to be isolated from complex routing and policy management within the standard user space. Applications which work against the default interface can be run against a specific wlan interface by running them within the namespace. Running in this isolated context would not disrupt the ethernet interface from being the primary route in userspace.

### How would this be implemented
The implementation would transition to the manual management of the wpasupplicant configuration file to ensure all modes and features of the wpasupplicant can be utilised. the namespace approach allows multiple instances of wpasupplicant to run simultaneously by running one in each namespace as needed, each with its own managed configuration file.

The alternative would be to manage routing tables and policies and although possible, could give rise to accidental testing of the wrong interface. **Isolation through namespaces is more robust as it can be engineered so only one interface is available and this will be tested**. It additionally allows multiple wlan adapters to connect to the same SSID for purposes such as testing the impact of multi-client connectivity - e.g., observing client counts change, looking at the impact of airtime utilisation, etc.

The proposal would be to migrate the DBUS implementation within core to a namespace implementation which manages separate wpasupplicant configurations in each namespace. This will provide persistence and isolation. Bridging can be investigated at a later date if needed.

### Minimal changes to the API
The existing API would remain broadly the same. Additional control would be added to specify whether a connection should be made in a new namespace or not. 

- Making a connection in the userspace will still be possible, modifying wpasupplicant as is done today in the standard demos
- Requesting a connection is made in a separated context would be presented to the end user as 'isolated for testing client connections'. this would move one or more wlan interfaces to the namespace where it si configured and connected

### Additional considerations
Many existing applications will not handle namespaces so it must be possible to restore the connection back to the standard user space. E.g., if an interface is required for remote packet capture in monitor mode, then there will need to be a mechanism to ensure the adapter appropriately configured. Some additional API calls will be needed to obtain a readout of the 'mode' adapters are in - this could be 'isolated for testing' or 'generally available'. APIs to move and restore the connections need to be implemented.

### Self Service

- [x] I would be willing to fix this feature myself.

### Guidelines and Policies

- [x] I have read the contributing guidelines and agree to follow the code of conduct and contribution policies.

bentumbler, 2025-07-16:
For reference, the basic approach to moving the wlan adapter to a new namespace is as follows.
`sudo ip netns add wifi-ns
sudo iw phy phy0 set netns name wifi-ns
sudo ip netns exec wifi-ns bash
ip link set wlan0 up`

joshschmelzle, 2026-11-05:
Implemented as the `/network/config/*` namespace profiles (#113, #302, #405, #406), documented in `NETWORK_CONFIG.md`. Open defects have their own issues: #237, #312, #343, #358, #451.

closed 2026-11-05 by joshschmelzle